Spyro-Soft/scargo #52: [flash] Add port name option

icesar-spyro, 2023-02-10:
scargo flash --port <port name e.g. /dev/ttyUSB0>

jze-spyro, 2023-02-17:
I've started doing it, but I can't change the status, because I don't have write access yet

jze-spyro, 2023-02-17:
It seems that it should also work for stm32, but our current setup can't be extended with this option, and I don't feel competent to change the setup.

fkuatspyro, 2023-02-28:
Partially done (esp only) with version 1.0.7.

aaksenczuk, 2023-05-03:
port must be added in the following way to work with esptool:
**esptool.py --chip esp32 --port=/dev/ttyUSB0** write_flash --flash_mode dio --flash_freq 80m --flash_size 4MB 0x1000 bootloader/bootloader.bin 0x10000 esp32-ihub-scargo.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin                                                     
esptool.py v3.3.2
Serial port /dev/ttyUSB0
Connecting.......
Chip is ESP32-D0WD-V3 (revision 3)

aaksenczuk, 2023-05-04:
> @aaksenczuk This should be a new issue. This issue remains open, because we haven't added the port option for stm32. Since now you're making this about a fix for esp, are you going to close it when this issue is fixed?
> 
> By the way, from the error message it seems the problem is related to parttool.py rather than esptool.py (maybe both).

Actually present PR(#229) is covering albo the stm32 port management:
(Update port command to work with stm32 and esp32:
stm32:
stlinkv2 command line: ./st-flash [--debug] [--reset] [--serial ] [--format ] [--flash=] {read|write}
stlinkv2 command line: ./st-flash [--debug] [--serial ] erase)

if this is not a problem I will keep it as single task managing the port

rap-spyro, 2023-05-23:
Fixed in #229

closed 2023-05-23 by rap-spyro