Spyro-Soft/scargo #100: Multiline copyrights not supported for check and fix

fkuatspyro, 2023-03-01:
**Describe the bug**
Multi-line copyrights are not supported by check and fix command.

**To Reproduce**
1. Create empty .h file.
2. Add multiline copyright in toml and run scargo update
```
[check.copyright]
description = """ 
Copyright 
this is 
multi-line copyright
"""
exclude = []
```
3. run `scargo fix --copyright` ->Adds invalid copyright
4. run `scargo check --copyright` -> Still reports error for copyright 

**Expected behavior**
Scargo fix --copyright adds valid copyright when and scargo check validates it correctly.

BohunSpyro, 2023-03-16:
scargo doesn't take updated copyrights from scargo.toml at all when command _scargo fix --copyright_

BohunSpyro, 2023-03-17:
Fixed.

jze-spyro, 2023-03-17:
No, it is **not** fixed.

> Expected behavior
> Scargo fix --copyright adds valid copyright when **and scargo check validates it correctly**.

Currently `scargo check` does not validate multiline copyright correctly.

closed 2023-04-17 by jze-spyro