Spyro-Soft/scargo #99: Improve output for check command

fkuatspyro, 2023-03-01:
**Is your feature request related to a problem? Please describe.**
Scargo checks do not leave clear messages on what is wrong. E.g. when copyright description is missing in toml.

**Describe the solution you'd like**
Messages that make it obvious to the user what the issue is. Potentially, create nice summary when running multiple checks.

jze-spyro, 2023-03-01:
We could also look through all log messages and check them for language and clearness. I saw some messages with incorrect grammar, like `"Cannot used docker command inside the docker container."` in `scargo.commands.docker`.

jze-spyro, 2023-03-03:
Some inconsistencies in `scargo check` output:
* copyright check logs with INFO level if copyright is missing, other checkers use warnings for similar messages
* cyclomatic and cppcheck do not call `sys.exit(1)` if the check fails

Exiting as soon as one check fails may not be the best course of action though, because it prevents subsequent checks from running.

closed 2023-04-24 by jze-spyro