Spyro-Soft/scargo #117: Warning appearing on any command run

icesar-spyro, 2023-03-10:
**Describe the bug**
Warning appearing on any command run

/usr/lib/python3/dist-packages/requests/__init__.py:89: RequestsDependencyWarning: urllib3 (1.26.14) or chardet (3.0.4) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "

**To Reproduce**

Run a command in scargo

**Expected behavior**
No warnings appearing.

jze-spyro, 2023-03-15:
It seems that the above description is not exactly accurate:
* it only happens in projects created without `--no-docker` flag
* it doesn't affect **all** commands, for example `scargo version` and `scargo update` are not affected. The affected ones include `scargo check` and `scargo build`

jze-spyro, 2023-03-28:
It still appears somewhat randomly, but I have an idea what might be the root cause.

PR #119 (2023-03-16): upgrade dependencies
It should fix #117, according to a manual test on local machine.
PR #139 (2023-03-29): Drop docker-compose==1.29.2 requirement
Should fix #117.

There's no reason to require docker-compose as a Python dependency. We really don't need it installed in our docker images.

And a big reason not to do it is that Python implementation of docker-compose is outdated (it has been rewritten to Golang and only this version is being supported). Also, since it's old, it depends on old versions of other packages, which causes conflicts.