Spyro-Soft/scargo #118: scargo new atParser - fail

aaksenczuk, 2023-03-15:
**Describe the bug**
When scargo project name consist capital letters then command fails

**To Reproduce**
run
scargo new atParser

**Expected behavior**
We can use any name which consists of alphanumeric and _ and - characters. 

**Additional context**
aak@WS-CKMBTG3:~/rep/tmp$ scargo new atParser
scargo INFO: Generated scargo.toml file
scargo INFO: Generated /home/aak/rep/tmp/atParser/src/atparser.cpp
scargo INFO: Generated /home/aak/rep/tmp/atParser/src/CMakeLists.txt
scargo INFO: Initialized git repo
scargo INFO: Generated /home/aak/rep/tmp/atParser/.devcontainer/Dockerfile-custom
scargo INFO: Generated /home/aak/rep/tmp/atParser/.devcontainer/docker-compose.yaml
scargo INFO: Generated /home/aak/rep/tmp/atParser/.devcontainer/Dockerfile
scargo INFO: Generated /home/aak/rep/tmp/atParser/.devcontainer/devcontainer.json
scargo INFO: Generated /home/aak/rep/tmp/atParser/.devcontainer/.env
scargo INFO: Generated /home/aak/rep/tmp/atParser/CMakeLists.txt
scargo INFO: Generated /home/aak/rep/tmp/atParser/conanfile.py
scargo INFO: Generated /home/aak/rep/tmp/atParser/tests/conanfile.py
scargo INFO: Generated /home/aak/rep/tmp/atParser/.gitlab-ci.yml
scargo INFO: Generated /home/aak/rep/tmp/atParser/tests/CMakeLists.txt
scargo INFO: Generated /home/aak/rep/tmp/atParser/tests/ut/CMakeLists.txt
scargo INFO: Generated /home/aak/rep/tmp/atParser/tests/it/CMakeLists.txt
scargo INFO: Generated /home/aak/rep/tmp/atParser/tests/mocks/CMakeLists.txt
scargo INFO: Generated /home/aak/rep/tmp/atParser/README.md
/usr/lib/python3/dist-packages/requests/__init__.py:89: RequestsDependencyWarning: urllib3 (1.26.14) or chardet (3.0.4) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "
Building atParser_dev
[+] Building 0.0s (0/0)                                                                                                                                                                                 
WARNING: buildx: failed to get git commit: fatal: ambiguous argument 'HEAD': unknown revision or path not in the working tree.
Use '--' to separate paths from revisions, like this:
'git <command> [<revision>...] -- [<file>...]'
ERROR: invalid tag "atParser-dev:1.0": repository name must be lowercase
ERROR: Service 'atParser_dev' failed to build : Build failed

jze-spyro, 2023-03-15:
@aaksenczuk 
> any name which consists of alphanumeric and _ and - characters

So it is not required that the first character is a letter? Are `123`, `---`, `_-_-_`  all allowed project names?

aaksenczuk, 2023-03-16:
as we spoko finally we expecting [a-zA-z][a-zA-z0-9_-]*

PR #120 (2023-03-16): Allow capital letters in the project name
Fix #118
closed 2023-03-16 by aaksenczuk