Spyro-Soft/scargo #134: Generated unit tests are not automatically added to the test build

kmc-spyro, 2023-03-27:
**Describe the bug**
When the UT for some folder are created they are not automatically added to the test build

**To Reproduce**
1. Create a project with some simple library
```bash
scargo new some03
cd some03
.... create a library
```
2. Add some compiler warning to the source file in order to verify the file is compiled
- project in the attachment
[ut_gen.zip](https://github.com/Spyro-Soft/scargo/files/11080020/ut_gen.zip)

3. Open project in docker
```bash
scargo docker run
```

4. Build the project
```bash
scargo build
```
The compiler warning present in the build log

5. Build the tests
```bash
scargo test
```
No compile warning as no UT created for the library yet

6. Generate unit tests for the library
```bash
scargo gen -u src/libs/test_lib
```
Unit test project (cmake + UT source) created

7. Try to build the tests again
```bash
scargo test
```
The source file is not compiled

**Expected behavior**
Compiler warning visible in the build log indicating the source file is compiled

**Additional context**
Probably not all CMakeLists.txt files are updated in order to include the generated UT in the test build.

jas-spyro, 2023-04-12:
@jze-spyro or @aaksenczuk do you know if this is also not a bug? Are you sure in this CMakeLists.txt file the path should be absolute?
[image]

jze-spyro, 2023-04-12:
> @jze-spyro or @aaksenczuk do you know if this is also not a bug? Are you sure in this CMakeLists.txt file the path should be absolute?

It's a bug, absolute paths will cause problems for teams

closed 2023-04-13 by jze-spyro