Spyro-Soft/scargo #154: `cxxstandard` is not respected for tests

rap-spyro, 2023-04-05:
**Describe the bug**
`cxxstandard` in scargo.toml is not interpreted correctly for tests.

**To Reproduce**
#. Add `cxxstandard = "17"` to scargo.toml under `[tests]`.
#. `scargo update`
**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional context**
Add any other context about the problem here.

jze-spyro, 2023-04-07:
@rap-spyro 
> Add cxxstandard = "17" to scargo.toml under [tests].

Is it required that tests have a separate cxxstandard setting? Currently they don't, which can be seen in the documentation: https://spyro-soft.github.io/scargo/scargo/scargo-toml.html#tests

Maybe it's enough to reuse cxxstandard defined under `[project]` for tests?

rap-spyro, 2023-04-07:
It is possible that for the embedded target we would like to set a different setting, but I don't think it's needed at this point. Main issue is that it's not set for tests at all for now, so if you have a fix for that - reusing it would be enough.

closed 2023-04-07 by jze-spyro