Spyro-Soft/scargo #187: Easy crosscompile build between esp32, stm32 and x86

aaksenczuk, 2023-04-17:
**Is your feature request related to a problem? Please describe.**
-recompile the basic library (with no dependencies to particular hw) only change of the target in scargo.toml is needed
-When changing the target from stm32 to esp32 there are missing data in toml which need to be set manually. This cause problem with multiplatform build

**Describe the solution you'd like**
 1) scargo.toml file always contains data for all targets. As the targets are split into sections, 
1.5) there is a possibility to add --target flag to build to crosscompile between targets
2) Ci/CD support build and release for mutliplatform 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context about the feature request here.

jze-spyro, 2023-04-17:
@aaksenczuk 
I don't think there can be an easy way to crosscompile:
* all targets use some of the same file locations, but generated differently, so a `scargo update` will be needed for switching targets
* esp32 keeps source in `main` directory instead of `src`, how do you think we should work around it?

KonradTajstra, 2023-06-26:
Have anything been clarified from "Open points" since last update? e.g. "how to store/manage conan profiles"?

closed 2023-09-08 by aaksenczuk