Spyro-Soft/scargo #188: there should be a base docker images release for each target

aaksenczuk, 2023-04-17:
**Is your feature request related to a problem? Please describe.**
When scargo is released there should be a build docker image release for each target separately. then we can use this base target docker image to download it from the docker registry instead of building it locally and guarantee the same base for the release for everyone.

**Describe the solution you'd like**
The "base" target image in my opinion can be a part of the beginning of the docker file to the custom user layer. It should be separately deployed each time for each target (esp32, stm32, x86, ...) and released with each scargo release. The versioning of the released images should be the same as the versions of the scargo. 

e.g. 
###
# @copyright Copyright (C) 2022 Spyrosoft Solutions. All rights reserved.
# THIS FILE WAS GENERATED AUTOMATICALLY. DO NOT CHANGE IT.
###

ARG DOCKER_IMAGE_ROOT=scargo_esp32:1.2.0
FROM ${DOCKER_IMAGE_ROOT} as base_esp32
....

or

ARG DOCKER_IMAGE_ROOT=scargo_stm32:1.2.0
FROM ${DOCKER_IMAGE_ROOT} as scargo_stm32

or 

ARG DOCKER_IMAGE_ROOT=scargo_x86:1.2.0
FROM ${DOCKER_IMAGE_ROOT} as scargo_x86
...

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context about the feature request here.

jze-spyro, 2023-04-27:
^ mentioned by mistake, ignore it

aaksenczuk, 2023-10-17:
not to implement,

closed 2023-10-17 by aaksenczuk