Spyro-Soft/scargo #189: There is a possibility to define conan profile for each target seperatelly

aaksenczuk, 2023-04-17:
**Is your feature request related to a problem? Please describe.**
there should be a possibility to define conan profile for our targets (esp32, stm32, x86,...). Defining the target at the end should allow us to cross-compile our code for various platforms and using scargo publish - deploy it to the conan repo. Then we   will be able to download the library for the defined target

**Describe the solution you'd like**
taking data from scargo.toml we are able to define default conan profile for chosen target. we should be able to build using conan profile values instead of cmake directly 

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context about the feature request here.

kmc-spyro, 2023-04-17:
For instance I have tried the following profile to cross compile a library for esp32 (out of IDF build):
```
[settings]
os=baremetal
os_build=Linux
arch=xtensalx6
arch_build=x86_64
compiler=gcc
compiler.version=8
compiler.libcxx=libstdc++
build_type=Debug
[options]
[build_requires]
[env]
CMAKE_TOOLCHAIN_FILE=/opt/esp-idf/tools/cmake/toolchain-esp32.cmake
```

jze-spyro, 2023-04-18:
This is blocked by #187, right?

kmc-spyro, 2023-04-25:
This profile I have tried for STM32:
```
toolchain_path=/opt/gcc-arm-none-eabi/bin
toolchain=arm-none-eabi
[settings]
os=baremetal
os_build=Linux
arch=armv7
arch_build=x86_64
compiler=gcc
compiler.version=10
compiler.libcxx=libstdc++
build_type=Release
[options]
[build_requires]
[env]
STM32_CHIP=STM32L496AG
STM32_TOOLCHAIN_PATH=/opt/gcc-arm-none-eabi
STM32_TARGET_TRIPLET=arm-none-eabi
CMAKE_TOOLCHAIN_FILE=/workspace/third-party/stm32-cmake/cmake/stm32_gcc.cmake
```

closed 2023-07-26 by KonradTajstra