Spyro-Soft/scargo #204: Allow for manual process calls override

JakubZurkowski-spyro, 2023-04-25:
**Is your feature request related to a problem? Please describe.**
Sometimes I need to call a process used by scargo with additional arguments or exchange it with my own tool.

**Describe the solution you'd like**
`scargo.toml` should allow for such process call substititution (preferably as a template string filled with scargo variables).

**Describe alternatives you've considered**
If whole process substitution is not possible then at least it should be possible to provide additional arguments used for process calls (e.g. `cppcheck --inline-suppr`).

jze-spyro, 2023-04-27:
@aaksenczuk 
I'm not sure if we should do something like that.

jze-spyro, 2023-04-28:
In most cases it won't be possible to just replace the command, because it may have unintended consequences, especially substituting a completely different tool. We might have a plugin system for adding other tools, but we don't and it would be a lot of work to add it.
For use cases it may make more sense to just add whatever config variables are needed. `--inline-suppr` was added unconditionally, but some other, more obtrusive changes may be added guarded by appropriate config variables.
I suggest closing it as not planned.

JakubZurkowski-spyro, 2023-05-02:
This issue is partly about the project philosophy - should scargo enforce use of specific set of tools or should it be a customizable framework?

@aaksenczuk reminded me that such changes can always be introduced by inheriting the current codebase and tinkering with it manually. That's good enough for now.

aaksenczuk, 2023-05-03:
I will rather not go with direct call and API extension of scargo.
Propose a solution: add an additional option to scargo.toml file related to the changes. We will be sure then that everyone are using same set of rules over the whole project. For this I will suggest to precisely define what can be modify.
Alternative is to exclude .clang-tidy hidden file in scargo.toml and do not put any further modifications

closed 2023-05-26 by jze-spyro