Spyro-Soft/scargo #205: [CI] build step is missing conan initialization

JakubZurkowski-spyro, 2023-04-25:
**Describe the bug**
When I put my project in GitLab then the build stage fails because of missing dependencies.

```shell
$ scargo build
...
Configuration:
[settings]
arch=x86_64
arch_build=x86_64
build_type=Release
compiler=gcc
compiler.libcxx=libstdc++
compiler.version=9
os=Linux
os_build=Linux
[options]
[build_requires]
[env]
conanfile.py (scargo-stm32-blink/0.1.0): Installing package
Requirements
Packages
Installing (downloading, building) binaries...
conanfile.py (scargo-stm32-blink/0.1.0): Generator txt created conanbuildinfo.txt
conanfile.py (scargo-stm32-blink/0.1.0): Generator cmake created conanbuildinfo.cmake
conanfile.py (scargo-stm32-blink/0.1.0): Aggregating env generators
conanfile.py (scargo-stm32-blink/0.1.0): Generated conaninfo.txt
conanfile.py (scargo-stm32-blink/0.1.0): Generated graphinfo
CMake Error at CMakeLists.txt:20 (include):
  include could not find requested file:
    third-party/stm32-cmake/cmake/stm32/common.cmake
CMake Error at CMakeLists.txt:21 (stm32_get_chip_info):
  Unknown CMake command "stm32_get_chip_info".
-- Configuring incomplete, errors occurred!
scargo ERROR: Unable to build exec file
...
```

**Expected behavior**
A generated `.gitlab-ci.yml` should include `conan source .` command before `scargo build`. Some projects may not need it so this should be optional (set in `scargo.toml`).

**Additional context**
`conan source .` is currently part of `conan update` which is skipped in CI builds. Nevertheless, this command seems critical for CI building.

stanik-spyro, 2023-08-08:
Conan implementation in scargo is currently under full rewrite process (versions 1.6.1, 1.6.2, 1.6.3 of scargo). 
Before them it safe to say conan was completly broken (yes - sentence like "we have conan support" was wrong from the start)
This issue should be closed as reproduction of bug is no longer possible.

KonradTajstra, 2023-08-08:
scargo build could be potentially enhanced by adding scargo source at start - to be refined. In conan documentation it is advised to use these three steps together: source, install, build - https://docs.conan.io/1/reference/commands/development/build.html

KonradTajstra, 2023-08-10:
> scargo build could be potentially enhanced by adding scargo source at start - to be refined. In conan documentation it is advised to use these three steps together: source, install, build - https://docs.conan.io/1/reference/commands/development/build.html

conan source is now called in scargo build, so the problem should be solved https://github.com/Spyro-Soft/scargo/pull/318

closed 2023-08-10 by KonradTajstra