Spyro-Soft/scargo #211: Clang-format run locally reports different errors than on the CI

mdk-spyro, 2023-04-26:
**Describe the bug**
Local scargo check --clang-format reports different errors than a CI job running the same command. Running scargo fix on the local machine will correct formatting for the errors found locally but not for the errors found on CI. Also issues corrected by local scargo fix can then be indicated as errors on CI. This makes a formatting issues to be corrected manually what is not trivial sometimes.

**To Reproduce**
Run scargo check --clang-format on your local machine and then push the changes to the GitLab repository and run "Job clang format" job on CI (if not started automatically).

**Expected behavior**
Running scargo check --clang-format should always give the same results no matter if it is run locally or on the CI. Issues corrected by scargo fix locally cannot be indicated as errors by the scargo check --clang-format run on CI.

jze-spyro, 2023-04-26:
@mdk-spyro
1. Are you running scargo in docker or natively? What's the `build-env` setting in scargo.toml?
2. If you're running scargo natively, what's the result of `scargo check --clang-format` inside a docker container? Is it more like what happens in CI?
3. What is your distribution/version (e.g. Ubuntu 22.04, Debian bullseye etc.)?
4. Can you give some examples of the problems you're experiencing?

CI uses Ubuntu 20.04 by default (you can change it by editing the `DOCKER_IMAGE_ROOT` variable in `.devcontainer/.env`). If you have a different distribution/version locally, it will probably have a different version of `clang-format` which may work differently.

mdk-spyro, 2023-04-26:
1. As far as I know scargo will run docker container to execute the command in the docker so it shouldn't matter, or am I wrong? I have a `build-env = "docker"` in scargo.toml.
2. Locally it does not matter if I simply run `scargo check --clang-format` or I firstly run `scargo docker run` and then `scargo check --clang-format` - the results are the same.
3. Locally I use Ubuntu 22.04.2 LTS and as I can see the docker image is based on the same Ubuntu 22.04.
4. For example currently the CI job for clang-format check on the project I am working on is passing without any issue while locally I have the following errors (example):
`main/tasks/ota_task/ota_task.cpp:27:49: error: code should be clang-formatted [-Wclang-format-violations]
    iot_hub_.register_desired_property_callback(
                                                ^
main/tasks/ota_task/ota_task.cpp:28:24: error: code should be clang-formatted [-Wclang-format-violations]
        "deviceUpdate",
                       ^
main/tasks/ota_task/ota_task.cpp:29:64: error: code should be clang-formatted [-Wclang-format-violations]
        [this](const std::string& property, const cJSON* value) { this->onDesiredPropertyChange(property, value); });
                                                               ^
scargo INFO: 
main/connectivity/fota_agent/fota_agent.cpp:198:59: error: code should be clang-formatted [-Wclang-format-violations]
            [&](int chunk_number, std::string& chunk_data) {
                                                          ^
scargo INFO: 
main/connectivity/network_connection/cellular_connection/src/cellular_connection.cpp:472:18: error: code should be clang-formatted [-Wclang-format-violations]
    auto status = std::any_of(&pdn_status_buffer[0], &pdn_status_buffer[status_count - 1], [&](const auto& pdn_status) {
                 ^
main/connectivity/network_connection/cellular_connection/src/cellular_connection.cpp:472:91: error: code should be clang-formatted [-Wclang-format-violations]
    auto status = std::any_of(&pdn_status_buffer[0], &pdn_status_buffer[status_count - 1], [&](const auto& pdn_status) {
                                                                                          ^
main/connectivity/network_connection/cellular_connection/src/cellular_connection.cpp:472:119: error: code should be clang-formatted [-Wclang-format-violations]
    auto status = std::any_of(&pdn_status_buffer[0], &pdn_status_buffer[status_count - 1], [&](const auto& pdn_status) {
                                                                                                                      ^
main/connectivity/network_connection/cellular_connection/src/cellular_connection.cpp:472:121: error: code should be clang-formatted [-Wclang-format-violations]
    auto status = std::any_of(&pdn_status_buffer[0], &pdn_status_buffer[status_count - 1], [&](const auto& pdn_status) {
`
If I run `scargo fix` and push the changes then the CI job will fail indicating those corrections as errors. It behave like there are two different versions of clang-format config file.

jze-spyro, 2023-04-26:
@mdk-spyro 

What is the first (`image:`) line of your `.gitlab-ci.yml.j2` and is it consistent with the `docker-image-tag` setting in `scargo.toml`? Does the CI worker have access to an up-to-date image with this tag?

Btw, please use triple backticks for code blocks (I've fixed that for you).

mdk-spyro, 2023-04-27:
Image param is set to: `image: harbor.spyrosoft.it/cdf/esp32-ihub-scargo-dev:1.4` and is the same as the one set in scargo.toml. Potentially there still could be differences between images as the CI uses the image downloaded from the harbor while locally docker image is build by `scargo update` and may install newer packages then those uploaded to harbor. But I do not think that this is the case in this situation because I have uploaded an image built locally to the harbor recently.

jze-spyro, 2023-04-27:
@mdk-spyro 
> My native env is Ubuntu 22.04.2 LTS and as I can see the docker image is based on the same Ubuntu 22.04.

I've downloaded the image and I can see that it's 20.04
```
$ docker run -it harbor.spyrosoft.it/cdf/esp32-ihub-scargo-dev:1.4 cat /etc/os-release | grep PRETTY_NAME
PRETTY_NAME="Ubuntu 20.04.5 LTS"
```

But it turns out it doesn't really matter, I wasn't aware that `esp-idf` ships its own version of `clang-format`:

```
mdk@8885d97205f1:/workspace$ which clang-format
/opt/esp-idf/tools/xtensa-clang/12.0.1-d9341b81fc/xtensa-esp32-elf-clang/bin/clang-format
```

Now, both locally and in CI clang-format is being called the same way, with `scargo check --format`. As far as I understand, they should **both** call the xtensa version, but for some reason it doesn't happen. Can you attach the full log from the clang format job on CI?

mdk-spyro, 2023-04-27:
1. I am sorry, I double checked the Ubuntu version and you are right that this is 20.04 and not 22.04. My mistake.
2. Please find the log from clang-format job on the CI:
```
Running with gitlab-runner 14.9.1 (bd40e3da)
  on gitlab-runner-gitlab-runner-585459c8dc-th7pw ZbUKWbNz
Preparing the "kubernetes" executor 00:00
Using Kubernetes namespace: gitlab-runners
Using Kubernetes executor with image harbor.spyrosoft.it/cdf/esp32-ihub-scargo-dev:1.4 ...
Using attach strategy to execute scripts...
Preparing environment 00:04
Waiting for pod gitlab-runners/runner-zbukwbnz-project-228-concurrent-0fsbzx to be running, status is Pending
Running on runner-zbukwbnz-project-228-concurrent-0fsbzx via gitlab-runner-gitlab-runner-585459c8dc-th7pw...
Getting source from Git repository 00:01
Fetching changes with git depth set to 20...
Initialized empty Git repository in /builds/iot/esp32-ihub-scargo/.git/
Created fresh repository.
Checking out ce1a414a as refs/merge-requests/68/head...
Updating/initializing submodules recursively with git depth set to 20...
Executing "step_script" stage of the job script 00:04
$ scargo check --clang-format
scargo INFO: Starting clang-format check...
scargo INFO: Skipping main/third_party/cellular_socket_tls/pack_struct_end.h
scargo INFO: Skipping main/third_party/cellular_socket_tls/pack_struct_start.h
scargo INFO: Skipping main/third_party/cellular_socket_tls/mbedtls_freertos_port.h
scargo INFO: Skipping main/third_party/cellular_socket_tls/FreeRTOSIPConfig.h
scargo INFO: Skipping main/third_party/cellular_socket_tls/using_mbedtls.h
scargo INFO: Skipping main/third_party/cellular_socket_tls/mbedtls/esp_config.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/modem_BG96/inc/cellular_bg96.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/modem_BG96/inc/cellular_bg96_setup.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/modem_A7682E/inc/cellular_a7682e_setup.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/modem_A7682E/inc/cellular_a7682e.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/cellular_interface/interface/cellular_comm_interface.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/cellular_interface/include/cellular_types.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/cellular_interface/include/cellular_api.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/cellular_interface/include/cellular_config_defaults.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/cellular_interface/include/common/cellular_common_portable.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/cellular_interface/include/common/cellular_at_core.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/cellular_interface/include/common/cellular_common_api.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/cellular_interface/include/common/cellular_common.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/cellular_interface/include/private/cellular_pktio_internal.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/cellular_interface/include/private/cellular_internal.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/cellular_interface/include/private/cellular_common_internal.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/cellular_interface/include/private/cellular_pkthandler_internal.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/cellular_interface/helper/cellular_platform.h
scargo INFO: Skipping main/third_party/freertos-cellular-port/cellular_interface/helper/cellular_config.h
scargo INFO: Skipping main/third_party/networkInterfaceTls/pack_struct_end.h
scargo INFO: Skipping main/third_party/networkInterfaceTls/dev_logger_cfg.h
scargo INFO: Skipping main/third_party/networkInterfaceTls/dev_logger.h
scargo INFO: Skipping main/third_party/networkInterfaceTls/pack_struct_start.h
scargo INFO: Skipping main/third_party/networkInterfaceTls/logging_levels.h
scargo INFO: Skipping main/third_party/networkInterfaceTls/logging_stack.h
scargo INFO: Skipping main/third_party/networkInterfaceTls/cellular_config.h
scargo INFO: Skipping main/third_party/networkInterfaceTls/FreeRTOSIPConfig.h
scargo INFO: Skipping main/third_party/coreMqtt/demo_config.h
scargo INFO: Skipping main/third_party/coreMqtt/azure_iot_config.h
scargo INFO: Skipping main/third_party/coreMqtt/core_mqtt_config.h
scargo INFO: Skipping main/third_party/azure-iot/backoff_algorithm.h
scargo INFO: Skipping main/third_party/azure-iot/common/transport/sockets_wrapper.h
scargo INFO: Skipping main/third_party/azure-iot/common/transport/transport_tls_socket.h
scargo INFO: Skipping main/third_party/azure-iot/common/transport/transport_abstraction.h
scargo INFO: Skipping main/third_party/azure-iot/common/transport/transport_socket.h
scargo INFO: Skipping main/third_party/azure-iot/common/utilities/azure_sample_crypto.h
scargo INFO: Skipping main/third_party/azure-iot/common/azure_ca_recovery/azure_ca_recovery_storage.h
scargo INFO: Skipping main/third_party/azure-iot/common/azure_ca_recovery/azure_ca_recovery_rsa_verify.h
scargo INFO: Skipping main/third_party/azure-iot/common/azure_ca_recovery/azure_ca_recovery_parse.h
scargo INFO: Skipping main/third_party/cellular_socket_tls/mbedtls_freertos_port.c
scargo INFO: Skipping main/third_party/cellular_socket_tls/using_mbedtls.c
[... cut ...]
scargo INFO: Skipping main/third_party/azure-iot/common/azure_ca_recovery/azure_ca_recovery_parse.c
scargo INFO: Skipping main/third_party/azure-iot/common/azure_ca_recovery/azure_ca_recovery_mbedtls_rsa_verify.c
scargo INFO: Finished clang-format check. Found problems in 0 files.
Uploading artifacts for successful job 00:01
Uploading artifacts...
WARNING: build/clang_format.log: no matching files 
ERROR: No files to upload                          
Cleaning up project directory and file based variables 00:00
Job succeeded
```

jze-spyro, 2023-04-27:
@mdk-spyro 

Thank you. I don't have any simple way to test gitlab-ci.yml, but I have an idea for a fix. You can try adding `source /opt/esp/entrypoint.sh` to the clang format script in your `.gitlab-ci.yml` like this:
```yaml
clang format:
  stage: check
  extends: .merge_request
  allow_failure: false
  script:
    - source /opt/esp/entrypoint.sh
    - scargo check --clang-format
  artifacts:
    paths:
      - build/clang_format.log
    when: always
```
If it works, I'll make a PR.

closed 2023-05-04 by jze-spyro