Spyro-Soft/scargo #223: Cannot call any command in docker with esp32 project

jobspyrosoft, 2023-04-28:
**Describe the bug**
Cannot call commands inside docker in esp32 project

**To Reproduce**
```
scargo new {project_name} --target esp32
cd {project_name}
scargo docker run
scargo update
```

**logs**
job@3bda65868a3a:/workspace$ scargo update
Traceback (most recent call last):

  File "/usr/local/bin/scargo", line 8, in <module>
    sys.exit(cli())

  File "/usr/local/lib/python3.8/dist-packages/scargo/cli.py", line 441, in update
    scargo_update(config_file_path)

  File "/usr/local/lib/python3.8/dist-packages/scargo/commands/update.py", line 47, in scargo_update
    config = get_scargo_config_or_exit(config_file_path)

  File "/usr/local/lib/python3.8/dist-packages/scargo/config_utils.py", line 32, in get_scargo_config_or_exit
    return parse_config(config_file_path)

  File "/usr/local/lib/python3.8/dist-packages/scargo/config.py", line 235, in parse_config
    return Config.parse_obj(config_obj)

  File "pydantic/main.py", line 526, in pydantic.main.BaseModel.parse_obj

  File "pydantic/main.py", line 339, in pydantic.main.BaseModel.__init__

  File "pydantic/main.py", line 1102, in pydantic.main.validate_model

  File "/usr/local/lib/python3.8/dist-packages/scargo/config.py", line 57, in validate_special_configs
    target_id = values["project"].target_id

KeyError: 'project'

aaksenczuk, 2023-05-03:
this issue is related to #225

aaksenczuk, 2023-05-03:
in my opinion `if not Path(project_path, ".dockerenv").exists()`:... should be update to 
   `if not Path(project_path, "**/**.dockerenv").exists():`

jze-spyro, 2023-05-04:
This is because the project was created using a local (unreleased) version of `scargo`, and docker installs the released version. There was a change for esp32 and stm32 targets and now the compiler settings are not generated for them, but they were required by the released version.
Other than that, the error message is wrong, I'm going to fix it.

jze-spyro, 2023-05-04:
I've fixed the error reporting issue (but until the next release the behavior won't change). Keep in mind that the released version does not contain the reported problem.