Spyro-Soft/scargo #225: Cannot run scargo update inside docker on stm32

jobspyrosoft, 2023-04-28:
**Describe the bug**
Cannot run scargo update inside docker on stm32. Got error FileNotFoundError: [Errno 2] No such file or directory: 'docker-compose', but file is existing in .devcontainer dir

**To Reproduce**
```
scargo new {project_name} --target stm32
scargo docker run
scargo update
```

**Logs**
```
job@69e3e15325b0:/workspace$ scargo update
/usr/local/lib/python3.8/dist-packages/requests/__init__.py:109: RequestsDependencyWarning: urllib3 (2.0.0) or chardet (None)/charset_normalizer (3.1.0) doesn't match a supported version!
  warnings.warn(
scargo INFO: Generated /workspace/.devcontainer/docker-compose.yaml
scargo INFO: Generated /workspace/.devcontainer/devcontainer.json
scargo INFO: Generated /workspace/.devcontainer/Dockerfile
scargo INFO: Generated /workspace/CMakeLists.txt
scargo INFO: Generated /workspace/conanfile.py
scargo INFO: Generated /workspace/tests/conanfile.py
scargo INFO: Generated /workspace/.gitlab-ci.yml
scargo INFO: Generated /workspace/tests/CMakeLists.txt
scargo INFO: Pulling the image from docker registry...
Traceback (most recent call last):

  File "/usr/local/bin/scargo", line 8, in <module>
    sys.exit(cli())

  File "/usr/local/lib/python3.8/dist-packages/scargo/cli.py", line 441, in update
    scargo_update(config_file_path)

  File "/usr/local/lib/python3.8/dist-packages/scargo/commands/update.py", line 102, in scargo_update
    if not pull_docker_image(docker_path):

  File "/usr/local/lib/python3.8/dist-packages/scargo/commands/update.py", line 111, in pull_docker_image
    result = subprocess.run(

  File "/usr/lib/python3.8/subprocess.py", line 493, in run
    with Popen(*popenargs, **kwargs) as process:

  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,

  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)

FileNotFoundError: [Errno 2] No such file or directory: 'docker-compose'
```

aaksenczuk, 2023-05-03:
cargo update actually is working inside the docker but do not update the docker itself (it will be then docker in docker). However, all file should be appropriately updated inside the docker. Propose a solution: we will skip the docker build/update step when the user is running scargo update from the docker so the error message will not pops up

rap-spyro, 2023-05-23:
Fixed in #233

closed 2023-05-23 by rap-spyro