Spyro-Soft/scargo #227: extend scargo.toml [test] section posibilities

aaksenczuk, 2023-05-02:
**Is your feature request related to a problem? Please describe.**
Add the possibility in for the test to add custom make flags as it is possible for the main project.

**Describe the solution you'd like**
There should be a possibility to define profiles for test and add custom flags as it is possible for main project: eg. 
[profile.Debug]
SIMU = "ON"
SDKCONFIG = "${CMAKE_BINARY_DIR}/sdkconfig"
Those flags then will be used in main UT cmake as a set()

jze-spyro, 2023-05-04:
There is nothing to be done, it already works like this. You can add custom cmake variables directly in the `[tests]` table in `scargo.toml`.
Personally I don't like it, because custom variables should not share a namespace with predefined ones (for example, any new predefined variable potentially breaks backwards compatibility), but I guess it's too late to change it.

aaksenczuk, 2023-05-04:
ok, to be tested

jze-spyro, 2023-05-25:
@aaksenczuk
has it been tested?

closed 2023-09-08 by aaksenczuk