Spyro-Soft/scargo #230: Getting errors while using scargo build in docker

jobspyrosoft, 2023-05-04:
**Describe the bug**
When using scargo build in docker there is error visible

```
/usr/local/lib/python3.8/dist-packages/requests/__init__.py:109: RequestsDependencyWarning: urllib3 (2.0.0) or chardet (None)/charset_normalizer (3.1.0) doesn't match a supported version!
  warnings.warn(
```

jze-spyro, 2023-05-04:
@jobspyrosoft 
You didn't include any reproduction steps. I'm running `scargo build` in docker and I'm not getting this message.

jobspyrosoft, 2023-05-04:
For me it's just like this:
```
scargo new my_project
cd my_project
scargo docker run
scargo build
```

jze-spyro, 2023-05-29:
I can't reproduce it. After the same steps I have `urllib3==1.26.16` installed, which doesn't cause this problem.

closed 2023-06-03 by aaksenczuk