Spyro-Soft/scargo #240: missleading error during esp32 debug

jobspyrosoft, 2023-05-10:
**Describe the bug**
missleading error during esp32 debug, in scargo docs we can see that debug isn't supported so this error during execution of scargo debug should be appropriate
for now it is:
`subprocess.CalledProcessError: Command '['xtensa-esp32-elf-gdb', PosixPath('/workspace/build/Debug/esp32debug.elf'), '--eval-command=target extended-remote localhost:3333']' returned non-zero exit status 127.`

**To Reproduce**
```
scargo new esp32project --target esp32
scargo docker run
scargo build --profile Debug
scargo debug
```

jze-spyro, 2023-05-25:
@jobspyrosoft 
> in scargo docs we can see that debug isn't supported

I guess you mean the "Scargo currently supports debug for x86 and stm32 targets" line in https://spyro-soft.github.io/scargo/scargo/scargo-debug.html, but you should have mentioned what you meant.

That line is outdated. Debugging for esp32 was added in March by @rap-spyro. So I don't think the error message is misleading, but it does seem that debugging doesn't work correctly for esp32.

jze-spyro, 2023-05-26:
The actual cause of the problem:

> xtensa-esp32-elf-gdb: error while loading shared libraries: libpython2.7.so.1.0: cannot open shared object file: No such file or directory

closed 2023-05-31 by jze-spyro