Spyro-Soft/scargo #245: Add posiiblity for easy generation of detailed code coverage report

stanik-spyro, 2023-05-16:
# Current state

The "scargo test" command generates by default a simple gcovr report (only coverage stats for files). The current state is good for CI statistic gathering as we get one simple file, but for eventual debugging and improvement of testing code, this is unuseful (You see "there is 50% coverage" but there is no information of what lines was not covered). Below is the current scargo code realizing the report (look at the --html parameter):

```
# Run code coverage.
  cmd = [
      "gcovr",
      "-r",
      "ut",
      ".",
      "-f",
      config.source_dir_path,
      "--html",
      "ut-coverage.html",
  ]
```

# Possible improvement description
Given above. The "--html" option forces simplified output. The detailed one is  "--html-details" but the latter will generate multiple files (which may not be nice for CI). So there are at least three options of improvement:

1. Change default to --html-details and change CI to recognize all files as artifacts.
2. Make report generation configurable at the "scargo test" command level to allow detailed report generation.
3. Generate both outputs. Short one for CI and statistics. Detailed one for developers searching bugs.

There is also possibility of other features like "output coverage as universal coverage format given as input to external tool"  eg. coveralls. Such things also require manipualtion with above gcovr options. But it might be another topic.

jze-spyro, 2023-05-25:
@rap-spyro Why is it a "good first issue" when it requires a non-trivial decision?

rap-spyro, 2023-05-26:
Adding an option for detailed report is trivial - it's just a matter of deciding between two gcovr parameters. Granted, the decision to change default behavior is more involved, but it's a separate topic than adding possibility for easy generation of detailed code coverage report which matters most for a developer.

jze-spyro, 2023-05-26:
Ok, I went with option 2