Spyro-Soft/scargo #260: Unexpected message during docker startup

aaksenczuk, 2023-06-03:
**Describe the bug**
scargo version: 1.6.0
I'm getting following message when entering the docker:
/usr/lib/python3/dist-packages/paramiko/transport.py:219: CryptographyDeprecationWarning: Blowfish has been deprecated
  "class": algorithms.Blowfish,

**To Reproduce**
1) scargo new presp32 --target esp32
2) scargo docker run

**Expected behavior**
This message should not pop up in the console during docker startup

KonradTajstra, 2023-09-05:
What pops up in 1.6.5 is:
```
/opt/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
```

KonradTajstra, 2023-09-13:
Fixed in https://github.com/Spyro-Soft/scargo/pull/347

closed 2023-09-13 by KonradTajstra