Spyro-Soft/scargo #283: scargo build command doesn't work for x86 platform image created with --lib argument

BartBart-pl, 2023-07-21:
**Describe the bug**
When user create new project and set --lib attribute, then it isn't possible to successfully finish scargo build command, below error is beeing thrown:

`CMake Error at src/CMakeLists.txt:9 (configure_file):
  Permission denied

ERROR: conanfile.py (proj_x86_1_6_1/0.1.0): Error in build() method, line 27
	cmake.configure()
	ConanException: Error 1 while executing cmake -G "Unix Makefiles" -DCMAKE_TOOLCHAIN_FILE="/workspace/build/Debug/conan_toolchain.cmake" -DCMAKE_INSTALL_PREFIX="/workspace/build/Debug/package" -DCMAKE_POLICY_DEFAULT_CMP0091="NEW" -DCMAKE_BUILD_TYPE="Debug" "/workspace"
scargo ERROR: Unable to build exec file
`

**To Reproduce**
1. Create new project with command:
`scargo new proj_x86_test --target=x86 --bin=hello_world --lib=test_lib`
Works ok 
2. Call 
`scargo docker update`
Works ok
3. Call
`scargo docker run`
Works ok
4. Call 
`scargo build` doesn't matter if called from docker or not

**Expected behavior**
Project should be build, and user should be able to call scargo run command which will show "Hello World" string. Building works for stm32 and esp32 targets

BartBart-pl, 2023-07-31:
Working fine on develop

KonradTajstra, 2023-08-08:
> Working fine on develop

Closing

closed 2023-08-08 by KonradTajstra