Spyro-Soft/scargo #285: Intergation tests concept

MichalMendrala, 2023-07-25:
**Is your feature request related to a problem? Please describe.**
Test scenarios need to be discussed, created and reviewed

**Describe the solution you'd like**
I decided to describe test scenarios here and after that discuss it and review with team.

**Describe alternatives you've considered**
Some page on Confluence or Wiki somewhere

**INTEGRATION TESTS TEST CONCEPT**

**General assumptions**

- tests will be organized as test suits, we will call them flows -  because they will be calling consecutive commands in defined order
- tests will be devided into three main groups depending of project type: backward compatibility tests (based on old projects stored in tests/test_data dir), new project bin option tests and new project lib option tests, those tests will be performed for all supported platforms (x86, esp32 and stm32) and for all aplicable scenarios

**TEST CASES**
Test cases are described in details under linked issues:

**Basic full flow** - basic flow containing as much commands as possible and covering the most frequent use case
- backward compatibility tests: 
- new project bin option tests:
- new project lib option tests:

**Special flows** - special flows to test specific command or specific use case, probably not always needs to be parametrized across all project types and platforms

KonradTajstra, 2023-07-26:
Would be nice to have integration test on two projects simultaneously. E.g. binary target for some architecture, let's say stm32, and second one with agnostic library (created for x86) that is used as a conan package. Can this be done internally in the lib using conans test_package?

MichalMendrala, 2023-09-12:
**Done:**
- backward compatibility tests: https://github.com/Spyro-Soft/scargo/issues/286
- new project bin option tests: https://github.com/Spyro-Soft/scargo/issues/290

**Still TO DO:**
- new project lib option tests: https://github.com/Spyro-Soft/scargo/issues/291
- test with agnostic library proposed by @KonradTajstra

closed 2023-12-23 by aaksenczuk