Spyro-Soft/scargo #290: [Test Scenario] Basic full flow new project bin only tests

MichalMendrala, 2023-07-26:
**Is your feature request related to a problem? Please describe.**
Need to define, discuss and review of test for new project with bin only

**Describe the solution you'd like**
Issue which will store test description

**Test case description**

For all platforms/targets (x86, esp32 and stm32)
1. Perform command: **new TEST_PROJECT_NAME --target TARGET --bin test_new_project**
2. Check if command finish with exit code == 0 (no error)
3. Check if following files and directories were created:
- ../TEST_PROJECT_NAME/CMakeLists.txt
- ../TEST_PROJECT_NAME/conanfile.py
- ../TEST_PROJECT_NAME/LICENSE
- ../TEST_PROJECT_NAME/README.md
- ../TEST_PROJECT_NAME/scargo.lock
- ../TEST_PROJECT_NAME/scargo.toml
- ../TEST_PROJECT_NAME/src(or main for esp32)/CMakeLists.txt
-  ../TEST_PROJECT_NAME/src(or main for esp32)/test_new_project.cpp
- ../TEST_PROJECT_NAME/tests/CMakeLists.txt
- ../TEST_PROJECT_NAME/tests/conanfile.py
- ../TEST_PROJECT_NAME/tests/it/CMakeLists.txt
- ../TEST_PROJECT_NAME/tests/ut/CMakeLists.txt
- ../TEST_PROJECT_NAME/tests/mocks/CMakeLists.txt
- ../TEST_PROJECT_NAME/tests/mocks/static_mock/CMakeLists.txt
- ../TEST_PROJECT_NAME/tests/mocks/static_mock/static_mock.h
5. Perform command: update
6. Check if command finish with exit code == 0 (no error)
7. Perform command: **docker run**
8. Check if command finish with exit code == 0 (no error)
9. If platform == esp32 coppy IDF file from /tests/test_data/test_projects/test_files/esp_32_idf_config/sdkconfig to copied project dir
10. Perform command: **update**
11. Check if command finish with exit code == 0 (no error) ??? should it run without error when docker is up ???
13. Perform command: **build**
14. Check if command finish with exit code == 0 (no error)
15. Check if dir ../build/Debug created
16. Perform command: **clean**
17. Check if command finish with exit code == 0 (no error)
18. Check if ../build dir was removed
19. Perform command: **build --profile Debug**
20. Check if command finish with exit code == 0 (no error)
21. Check if file ../build/Debug/bin/"project_name" was created
22. Perform command: **build --profile Release**
23. Check if command finish with exit code == 0 (no error)
24. Check if file ../build/Release/bin/"project_name" was created
25. Add new profile to toml file by: `add_profile_to_toml(
    "new",
    "cflags",
    "cxxflags",
    "cflags for new profile",
    "cxxflags for new profile",
)`
26. Perform command: **update**
27. Check if command finish with exit code == 0
28. Check if file .conan/profiles/x86_new containing expected "cflags" and "cxxflags" values
29. **Suggestion** maybe also change "cxxstandard" to see if CMakelits.txt was updated? or leave it for ut?
30. **Suggestion** maybe try to build using new profile **build --profile new**? 
31. Perform command: **gen --fs**
32. Check if command finish with exit code == 0 (no error)
33. Check if ../build/spiffs.bin file exist
34. Perform command: **gen -u path_to_src_dir** (path_to_src_dir="main" for esp32, path_to_src_dir="src" for stm32 and x86)
35. Check if command finish with exit code == 0 (no error)
36. Check if no file was created under tests/ut (only CMakeList.txt should be present in folder)
37. Create dummy lib_for_ut_gen_test.h flie in project src/main dir
38. Perform command: **gen -u path_to_src_dir** (path_to_src_dir="main" for esp32, path_to_src_dir="src" for stm32 and x86).
39. Check if command finish with exit code == 0 (no error)
40. Check "ut_lib_for_ut_gen_test.cpp" file exist uder ../tests/ut path
41. Perform command: **gen -m "path_to_h_file"**
42. Check if command finish with exit code == 0 (no error)
43. Check if following files was generated:
- ../tests/mocks/test_lib.h
- ../tests/mocks/mock_test_lib.h
- ../tests/mocks/mock_test_lib.cpp
44.  Perform command: **gen -c "device_id"** (dummy test device id)
45. Check if command finish with exit code == 0 (no error)
46. Check if following files was created:
- ../build/fs/
- ../build/fs/
- ../build/fs/
- ../build/certs/azure/
47. Perform command: **check**
48. Check if command finish with exit code == 0 (no error)
49. Perform command: **test**
50. Check if command finish with exit code == 0 (no error)
51. If target == x86: Perform command: **run**
52. If target == x86: Check if command finish with exit code == 0 (no error)
53. If target == x86: Check if "Hello World" in result output
54. Coppy files from  /tests/test_data/test_projects/test_files/fix_test_files to src dir
55. Perform command: **check**
56. Check if command finish with exit code == 1 (error not fixed files)
57. Perform command: **fix**
58. Check if command finish with exit code == 0 (no error now fixed files)
59. Check if following files from source dir containing following strings:
- fix_test_bin.cpp contains copyright text from toml file
- fix_test_lib.cpp contains copyright text from toml file
- fix_test_lib.h contains copyright text from toml file
- fix_test_lib.h contains "#pragma once" text
60. Perform command: **check**
61. Check if command finish with exit code == 0 (no error now fixed files)

What to do with comands:
- debug, doc, flash, publish

KonradTajstra, 2023-08-09:
Step 16 - What can be also checked is if "config/conan/profiles/<target>_<profile> file was created

KonradTajstra, 2023-08-09:
Step 18 - Suggestion: consider renaming to test_flags_stored_in_compile_commands_file - to be more verbose

KonradTajstra, 2023-08-09:
Step 23. and 25. - These steps should be executed in reversed order - first gen --certs, then gen --fs. This could be even combined in one step. For stm32 and x86 '--gen' is not supported so flow could be as described in step 25. But for esp32 I suggest following:
- execute step 25, check if certs are generated,
- add some dummyfile to main/fs
- check if build/fs contains dummyfile
- check spiffs.bin like in step 23

MichalMendrala, 2023-08-09:
> Step 18 - Suggestion: consider renaming to test_flags_stored_in_compile_commands_file - to be more verbose
> 
> I will put definition in tests

> Step 18 - Suggestion: consider renaming to test_flags_stored_in_compile_commands_file - to be more verbose

You are right, I will put definition of this flags.

MichalMendrala, 2023-08-09:
All suggestions implemented, thanks @KonradTajstra for great cooperation during realization of this task.

MichalMendrala, 2023-09-12:
Done

closed 2023-09-12 by MichalMendrala