Spyro-Soft/scargo #360: Conan remote repository not added

bkresaj, 2023-10-09:
**Describe the bug**
After running the docker container, conan remote repository set inside scargo.toml file is not visible inside docker container.

**To Reproduce**
scargo docker run
conan remote list

**Expected behavior**
conan remote list should show repository that was set inside scargo.toml file inside [conan.repo]

sho-spyrosoft, 2023-10-16:
The conan remote list will only updated when the project is build (scargo build --profile Release).
After that the added repository is available for conan remote list.

sho-spyrosoft, 2023-10-16:
Added conan_add_remote function to update.py.
Now the conan remotes will be applied by executing scargo update or scargo build

bkresaj, 2023-10-19:
Can you please add it also to scargo test? I would like to maybe only run scargo test inside docker container but remote conan repository is not added then, I have to add it manually.

closed 2023-10-31 by sho-spyrosoft