Spyro-Soft/scargo #372: Pull image error with scargo new command

aaksenczuk, 2023-10-16:
**Describe the bug**
When creating a new project (or updating an existing one) we get unexpected errors from docker image pulling that may be misleading for the user's suggestions that something went wrong when this is expected behavior. 

**To Reproduce**
create new project with scargo new command

**Expected behavior**
When creating a new project the error should not be visible as there is no chance that the image exists at this state.
For already existing project with scargo updates i suggest to change the warning message that " Docker image may not exist for pull" or something like that

**Additional context**
scargo INFO: Pulling the image from docker registry...
scargo WARNING:  stm_d1_dev Pulling 
 stm_d1_dev Warning 
WARNING: Some service image(s) must be built from source by running:
    docker compose build stm_d1_dev
1 error occurred:
        * Error response from daemon: pull access denied for stm_d1-dev, repository does not exist or may require 'docker login': denied: requested access to the resource is denied

sho-spyrosoft, 2023-10-17:
Which version do you use?

sho-spyrosoft, 2023-10-17:
Fixed this issue by remove check parameter from subprocess.run in function pull_docker_image.

closed 2023-10-17 by sho-spyrosoft