Spyro-Soft/scargo #380: Different mypy output from ci and pre-commit

fkuatspyro, 2023-10-23:
**Describe the bug**
mypy check on CI is passing, while mypy check in precommit hook is failing

**To Reproduce**
1. checkout develop
2. pre-commit run -a (setup pre-commit if not already)

**Expected behavior**
pre-commit hook and CI are running mypy in same way

**Additional context**
Add any other context about the problem here.

sho-spyrosoft, 2023-10-23:
In the CI the mypy won't be executed.
[image]

fkuatspyro, 2023-10-24:
No, this is not the reason as requirements.txt is generated using pip-compile `pip-compile --all-extras --output-file=ci/requirements.txt pyproject.toml`.
Same pacakges will be installed if you install using pip based on pyproject.toml `pip install .[dev,doc]` which is what I do.
[image]

closed 2023-10-31 by fkuatspyro