Spyro-Soft/scargo #397: Invalid default conan profiles for atsam target

kmc-spyro, 2023-11-15:
**Describe the bug**
The default project for atsam is not buildable after adding any dependency.

**To Reproduce**
* `scargo new --target atsam atsam-test`
* `cd atsam-test`
* Edit `scargo.toml` and add any dependency - e.g., `cjson/1.7.16`:
```yml
[dependencies]
#general -> public conan dependencies of project they will be added to package info (eg. bianry dynamic linkage libary) they will be added also to scargo test.
general = [
        "cjson/1.7.16"
]
```
* `scargo update`
* `scargo docker run`
* `scargo build`

**Expected behavior**
Build is scuccessful.

**Additional context**
The build failure:
```bash
======== Installing packages ========
cjson/1.7.16: Sources downloaded from 'conancenter'
cjson/1.7.16: Calling source() in /home/kmc/.conan2/p/cjson093bf0ae2099e/s/src

-------- Installing package cjson/1.7.16 (1 of 1) --------
cjson/1.7.16: Building from source
cjson/1.7.16: Package cjson/1.7.16:0cb11cd341acda7d8c01b4ccfcc80bbd11db11b2
cjson/1.7.16: Copying sources to build folder
cjson/1.7.16: Building your package in /home/kmc/.conan2/p/b/cjson57d0af2746f80/b
cjson/1.7.16: Calling generate()
cjson/1.7.16: Generators folder: /home/kmc/.conan2/p/b/cjson57d0af2746f80/b/build/Debug/generators
cjson/1.7.16: CMakeToolchain generated: conan_toolchain.cmake
cjson/1.7.16: CMakeToolchain generated: CMakePresets.json
cjson/1.7.16: CMakeToolchain generated: ../../../src/CMakeUserPresets.json
cjson/1.7.16: Generating aggregated env files
cjson/1.7.16: Generated aggregated env files: ['conanbuild.sh', 'conanrun.sh']
cjson/1.7.16: Calling build()
cjson/1.7.16: apply_conandata_patches(): No patches defined in conandata
cjson/1.7.16: Running CMake.configure()
cjson/1.7.16: RUN: cmake -G "Unix Makefiles" -DCMAKE_TOOLCHAIN_FILE="/home/kmc/.conan2/p/b/cjson57d0af2746f80/b/build/Debug/generators/conan_toolchain.cmake" -DCMAKE_INSTALL_PREFIX="/home/kmc/.conan2/p/b/cjson57d0af2746f80/p" -DCMAKE_POLICY_DEFAULT_CMP0042="NEW" -DCMAKE_POLICY_DEFAULT_CMP0091="NEW" -DCMAKE_BUILD_TYPE="Debug" "/home/kmc/.conan2/p/b/cjson57d0af2746f80/b/src"
-- Using Conan toolchain: /home/kmc/.conan2/p/b/cjson57d0af2746f80/b/build/Debug/generators/conan_toolchain.cmake
-- Conan toolchain: Setting CMAKE_POSITION_INDEPENDENT_CODE=ON (options.fPIC)
-- Conan toolchain: Setting BUILD_SHARED_LIBS = OFF
-- The C compiler identification is GNU 10.3.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc
-- Check for working C compiler: /opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc - broken
CMake Error at /usr/local/lib/python3.8/dist-packages/cmake/data/share/cmake-3.25/Modules/CMakeTestCCompiler.cmake:70 (message):
  The C compiler

    "/opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /home/kmc/.conan2/p/b/cjson57d0af2746f80/b/build/Debug/CMakeFiles/CMakeScratch/TryCompile-YpaKiS

    Run Build Command(s):/usr/bin/make -f Makefile cmTC_fa83f/fast && /usr/bin/make  -f CMakeFiles/cmTC_fa83f.dir/build.make CMakeFiles/cmTC_fa83f.dir/build
    make[1]: Entering directory '/home/kmc/.conan2/p/b/cjson57d0af2746f80/b/build/Debug/CMakeFiles/CMakeScratch/TryCompile-YpaKiS'
    Building C object CMakeFiles/cmTC_fa83f.dir/testCCompiler.c.obj
    /opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc   -Wall -Wextra -g  -fPIE -o CMakeFiles/cmTC_fa83f.dir/testCCompiler.c.obj -c /home/kmc/.conan2/p/b/cjson57d0af2746f80/b/build/Debug/CMakeFiles/CMakeScratch/TryCompile-YpaKiS/testCCompiler.c
    Linking C executable cmTC_fa83f
    /usr/local/lib/python3.8/dist-packages/cmake/data/bin/cmake -E cmake_link_script CMakeFiles/cmTC_fa83f.dir/link.txt --verbose=1
    /opt/gcc-arm-none-eabi/bin/arm-none-eabi-gcc -Wall -Wextra -g  CMakeFiles/cmTC_fa83f.dir/testCCompiler.c.obj -o cmTC_fa83f
    /opt/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/bin/ld: /opt/gcc-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/10.3.1/../../../../arm-none-eabi/lib/libg.a(lib_a-exit.o): in function `exit':
    exit.c:(.text.exit+0x2c): undefined reference to `_exit'
    collect2: error: ld returned 1 exit status
    make[1]: *** [CMakeFiles/cmTC_fa83f.dir/build.make:99: cmTC_fa83f] Error 1
    make[1]: Leaving directory '/home/kmc/.conan2/p/b/cjson57d0af2746f80/b/build/Debug/CMakeFiles/CMakeScratch/TryCompile-YpaKiS'
    make: *** [Makefile:127: cmTC_fa83f/fast] Error 2

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:4 (project)

```

kmc-spyro, 2023-11-15:
The build succeeds when toolchain file is created and used in the profile:

**atsam_Debug**
```yml
target_host=arm-none-eabi
standalone_toolchain=arm-none-eabi
cc_compiler=gcc
cxx_compiler=g++

[settings]
os=baremetal
arch=armv8
compiler=gcc
compiler.version=10
compiler.libcxx=libstdc++11
compiler.cppstd=gnu17
arch_build=x86_64
build_type=Debug

[env]
CONAN_CMAKE_FIND_ROOT_PATH=/opt/gcc-arm-none-eabi/lib/
#CC=$target_host-$cc_compiler
#CXX=$target_host-$cxx_compiler

[conf]
tools.build:cflags=["-Wall -Wextra -g"]
tools.build:cxxflags=["-Wall -Wextra -g"]
tools.cmake.cmaketoolchain:user_toolchain=["$PROFILE_DIR/arm-gcc-toolchain.cmake"]

[options]
[build_requires]
```

**arm-gcc-toolchain.cmake**
```cmake
set(CMAKE_SYSTEM_NAME Generic)
set(CMAKE_SYSTEM_PROCESSOR ARM)

set(TOOLCHAIN_PREFIX arm-none-eabi-)
find_program(BINUTILS_PATH ${TOOLCHAIN_PREFIX}gcc NO_CACHE)

if (NOT BINUTILS_PATH)
    message(FATAL_ERROR "ARM GCC toolchain not found")
endif ()

get_filename_component(ARM_TOOLCHAIN_DIR ${BINUTILS_PATH} DIRECTORY)
# Without that flag CMake is not able to pass test compilation check
if (${CMAKE_VERSION} VERSION_EQUAL "3.6.0" OR ${CMAKE_VERSION} VERSION_GREATER "3.6")
    set(CMAKE_TRY_COMPILE_TARGET_TYPE STATIC_LIBRARY)
else ()
    set(CMAKE_EXE_LINKER_FLAGS_INIT "--specs=nosys.specs")
endif ()

set(CMAKE_C_COMPILER ${TOOLCHAIN_PREFIX}gcc)
set(CMAKE_ASM_COMPILER ${CMAKE_C_COMPILER})
set(CMAKE_CXX_COMPILER ${TOOLCHAIN_PREFIX}g++)
set(CMAKE_AR ${TOOLCHAIN_PREFIX}gcc-ar)
set(CMAKE_RANLIB ${TOOLCHAIN_PREFIX}gcc-ranlib)

execute_process(COMMAND ${CMAKE_C_COMPILER} -print-sysroot
    OUTPUT_VARIABLE ARM_GCC_SYSROOT OUTPUT_STRIP_TRAILING_WHITESPACE)

# Default C compiler flags
set(CMAKE_C_FLAGS_DEBUG_INIT "-g3 -Og -Wall -pedantic -DDEBUG")
set(CMAKE_C_FLAGS_DEBUG "${CMAKE_C_FLAGS_DEBUG_INIT}" CACHE STRING "" FORCE)
set(CMAKE_C_FLAGS_RELEASE_INIT "-O3 -Wall")
set(CMAKE_C_FLAGS_RELEASE "${CMAKE_C_FLAGS_RELEASE_INIT}" CACHE STRING "" FORCE)
set(CMAKE_C_FLAGS_MINSIZEREL_INIT "-Os -Wall")
set(CMAKE_C_FLAGS_MINSIZEREL "${CMAKE_C_FLAGS_MINSIZEREL_INIT}" CACHE STRING "" FORCE)
set(CMAKE_C_FLAGS_RELWITHDEBINFO_INIT "-O2 -g -Wall")
set(CMAKE_C_FLAGS_RELWITHDEBINFO "${CMAKE_C_FLAGS_RELWITHDEBINFO_INIT}" CACHE STRING "" FORCE)
# Default C++ compiler flags
set(CMAKE_CXX_FLAGS_DEBUG_INIT "-g3 -Og -Wall -pedantic -DDEBUG")
set(CMAKE_CXX_FLAGS_DEBUG "${CMAKE_CXX_FLAGS_DEBUG_INIT}" CACHE STRING "" FORCE)
set(CMAKE_CXX_FLAGS_RELEASE_INIT "-O3 -Wall")
set(CMAKE_CXX_FLAGS_RELEASE "${CMAKE_CXX_FLAGS_RELEASE_INIT}" CACHE STRING "" FORCE)
set(CMAKE_CXX_FLAGS_MINSIZEREL_INIT "-Os -Wall")
set(CMAKE_CXX_FLAGS_MINSIZEREL "${CMAKE_CXX_FLAGS_MINSIZEREL_INIT}" CACHE STRING "" FORCE)
set(CMAKE_CXX_FLAGS_RELWITHDEBINFO_INIT "-O2 -g -Wall")
set(CMAKE_CXX_FLAGS_RELWITHDEBINFO "${CMAKE_CXX_FLAGS_RELWITHDEBINFO_INIT}" CACHE STRING "" FORCE)

set(CMAKE_OBJCOPY ${ARM_TOOLCHAIN_DIR}/${TOOLCHAIN_PREFIX}objcopy CACHE INTERNAL "objcopy tool")
set(CMAKE_SIZE_UTIL ${ARM_TOOLCHAIN_DIR}/${TOOLCHAIN_PREFIX}size CACHE INTERNAL "size tool")

set(CMAKE_SYSROOT ${ARM_GCC_SYSROOT})
set(CMAKE_FIND_ROOT_PATH ${BINUTILS_PATH})
set(CMAKE_FIND_ROOT_PATH_MODE_PROGRAM NEVER)
set(CMAKE_FIND_ROOT_PATH_MODE_LIBRARY ONLY)
set(CMAKE_FIND_ROOT_PATH_MODE_INCLUDE ONLY)
```

fkuatspyro, 2023-11-17:
Solved in Release 2.0.1

closed 2023-11-17 by fkuatspyro