Spyro-Soft/scargo #405: Regex rules for copyright check

kmc-spyro, 2023-12-13:
**Is your feature request related to a problem? Please describe.**
Required copyright statement format is fixed and not flexible. It requires to use standard c++ comments '//'.
And here we have a few problems:
- this could not work for other types of files, e.g. pure C
- doxygen cannot be used for copyrigth statement (`@copyrigth`)
- the lines of the statement cannot be flexible - for instance contain different year depending when the file was created

**Describe the solution you'd like**
Besides current approach also allow to use regex statement description in scargo.toml file, e.g.,:
```toml
[check.copyright]
description = '''
/\*\*
 \* @copyright Copyright \(C\) \d* SpyroSoft Solutions. All rights reserved.
 \*'''
exclude = []
```
and this would fit for instance this doxygen comment:
```cpp
/**
 * @copyright Copyright (C) 2023 SpyroSoft Solutions. All rights reserved.
 */
```

kmc-spyro, 2023-12-13:
Sample implementation in check.py:
```py
    def __get_copyright_lines_regex(self) -> List[str]:
        return [f"{el}" for el in self.copyright_desc.split("\n")] 

    def check_file(self, file_path: Path) -> CheckResult:
        copyright_lines = self.__get_copyright_lines()

        with open(file_path, encoding="utf-8") as file:
            file_lines = file.readlines()

            for line_no, line in enumerate(file_lines[: -(len(copyright_lines) - 1)]):
                if line == copyright_lines[0]:
                    if all(
                        line_from_file == line_from_copyrights
                        for line_from_file, line_from_copyrights in zip(
                            file_lines[line_no:], copyright_lines
                        )
                    ):
                        return CheckResult(problems_found=0)

        copyright_lines = self.__get_copyright_lines_regex()

        with open(file_path, encoding="utf-8") as file:
            file_lines = file.readlines()

            for line_no, line in enumerate(file_lines[: -(len(copyright_lines) - 1)]):
                if re.match(copyright_lines[0], line):
                    if all(
                        re.match(line_from_copyrights, line_from_file) != None
                        for line_from_file, line_from_copyrights in zip(
                            file_lines[line_no:], copyright_lines
                        )
                    ):
                        return CheckResult(problems_found=0)

        logger.warning("Missing copyright line in %s.", file_path)
        return CheckResult(problems_found=1)
```

aaksenczuk, 2023-12-15:
Just keep in mind that there can be any custom Copyright  set, not necessarily Spyro one

fkuatspyro, 2023-12-19:
Done in [release 2.1.1](https://github.com/Spyro-Soft/scargo/releases/tag/release_2.1.1)

closed 2023-12-19 by fkuatspyro