Spyro-Soft/scargo #419: unity support

spynolk, 2024-01-22:
Hi, would it be possible to add Unity support to scargo?
https://www.throwtheswitch.org/unity

aaksenczuk, 2024-01-23:
For now, our main focus is embedded systems. The main technology is here C/C++ in comparison to Unity where the main technology is C#. Unfortunately for now do do not have a plan to support unity

spynolk, 2024-01-23:
Just do avoid misunderstandings, I attach some information from the provided link:

> Unity is written in 100% pure C code. It follows ANSI standards while supporting most embedded compiler quirks.
> Unity is equally happy running tests for an 8-bit microcontroller as it is a 64-bit processor on steroids.

michalwloc, 2024-01-23:
Unity is a testing framework for embedded code written purely in C. Please make sure to verify the information in the attached link before closing the ticket: https://www.throwtheswitch.org/unity

aaksenczuk, 2024-01-23:
sorry for missunderstanding. Does unity has conan package version? if yes, you can simply try to add it into conan section in scargo.toml instead of gtest as in fact we are using ctest to manage test in scrago. can you try to find conan package, or create conan packer from unity?