Spyro-Soft/scargo #427: Leverage pyclean v3 features

bittner, 2024-05-27:
I saw you use (an older version of) pyclean in your `clean` module.

I'm not 100% sure what the module does in its entirety, but it's likely that the [clean function](https://github.com/Spyro-Soft/scargo/blob/main/clean.py#L61-L68) can be fully replaced by what PyClean provides today.

Instead of `py3clean`, which was dropped in pyclean v3, you might use something like:

```python
subprocess.run(f"pyclean . --debris --erase {*output_file_names} --verbose --dry-run".split())
```

The `--debris` option will take care of all commonly known build files and folders. With `--erase` you can clean up arbitrary files.

**Note:** Replace `--verbose --dry-run` by `--yes` once you're sure the code does what you expect. :bomb: 

Thanks for using PyClean! :cake: 

## References

- [Cleaning up arbitrary file with PyClean](https://github.com/bittner/pyclean?tab=readme-ov-file#clean-up-debris) (PyClean README)
- See example in [tox.ini](https://github.com/bittner/pyclean/blob/main/tox.ini#L25-L29) (PyClean project)

aaksenczuk, 2024-06-20:
thanks, will be added in the next release

bittner, 2024-07-03:
Note that your `pyproject.toml` file still references an old version of PyClean! I suggest you remove the version pinning or at least raise or to 3.0.0.